JAndrassy/ArduinoOTA #148: OTA with ETHERNET on OLIMEX ESP32 board / fail

E-Fabrice, 2022-04-04:
Hello,
I'm very interested by your ArduinoOTA library for a project.

My project use a OLIMEX ESP32 board (ESP32-POE).

I have made all recomendations into your installation process, removed
the legacy ESP32 OTA library, installed yours with the IDE.

then place the platform.local.txt into the right folder :

C:\Users\MYNAME\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.6

But i have an issue :

> In file included from C:\Program Files
> (x86)\Arduino\libraries\ArduinoOTA\examples\Advanced\OTASketchDownload\OTASketchDownload.ino:19:0:
> C:\Program Files(x86)\Arduino\libraries\ArduinoOTA\src/ArduinoOTA.h:127:22: error:
> 'WiFiServer' was not declared in this scope
>  ArduinoOTAMdnsClass <WiFiServer, WiFiClient, WiFiUDP> ArduinoOTA;
>                       ^
> C:\Program Files(x86)\Arduino\libraries\ArduinoOTA\src/ArduinoOTA.h:127:34: error:
> 'WiFiClient' was not declared in this scope
>  ArduinoOTAMdnsClass <WiFiServer, WiFiClient, WiFiUDP> ArduinoOTA;
>                                   ^
> C:\Program Files(x86)\Arduino\libraries\ArduinoOTA\src/ArduinoOTA.h:127:53: error:
> template argument 1 is invalid
>  ArduinoOTAMdnsClass <WiFiServer, WiFiClient, WiFiUDP> ArduinoOTA;
>                                                      ^
> C:\Program Files(x86)\Arduino\libraries\ArduinoOTA\src/ArduinoOTA.h:127:53: error:
> template argument 2 is invalid
> Plusieurs bibliothèque trouvées pour "WiFiUdp.h"
> Utilisé :
> C:\Users\BE\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.6\libraries\WiFi
> Non utilisé : C:\Program Files (x86)\Arduino\libraries\WiFi
> exit status 1
> Erreur de compilation pour la carte OLIMEX ESP32-EVB

above is the error message for a EVB card but i have same issue with the "POE" one ...

I already made a ethernet client on that board using the lib : ETH.h and i wonder if your library is compatible whith that ?
is there a workaround ??

Have you an idea please ?
Thanks in advance at all members for your help,
Best regards
Fabrice

jandrassy, 2022-04-04:
use the esp32 HTTPUpdate library. see its examples

E-Fabrice, 2022-04-04:
Hello Jan thanks you very much for this very quick reply, i'll try that ASAP and tell you !
good day
Fabrice

jandrassy, 2022-04-04:
> Hello Jan thanks you very much for this very quick reply, i'll try that ASAP and tell you ! good day Fabrice

My name is Juraj

E-Fabrice, 2022-04-04:
> My name is Juraj
my appologies ... thanks again Juraj

closed 2022-06-02 by JAndrassy