JAndrassy/ArduinoOTA #149: "Payload size wrong" with enc28j60

alexsandro-ss, 2022-04-06:
Hello,

Is the ArduinoOTA library compatible with enc28j60? I get "Payload size wrong" on the serial port when I use:

Serial.print("OTA upload response: ");
Serial.println(status);

at WiFiOTA.cpp

When I use W5500 everything works fine.

obs: I use enc28j60 with esp8266 (Wemos D1 MINI)

Thanks

jandrassy, 2022-04-06:
with EthernetENC or with the new lwIP_enc28j60?

alexsandro-ss, 2022-04-06:
I'm using EthernetENC with the basic OTEthernet example. I'm not using esp8266 Wifi. I connect via the Ethernet port of the ENC28J60.

I see it in the Arduino IDE:

'Connecting to board ...  done 
Uploading sketch ...  done
Flashing sketch ... Error flashing the sketch
Flashing sketch ... Error flashing the sketch'

My platform.local.txt

`tools.esptool.network_cmd={runtime.tools.arduinoOTA.path}/bin/arduinoOTA
tools.esptool.upload.network_pattern="{network_cmd}" -address {serial.port} -port 65280 -t 30 -username "Arduino" -password "password" -sketch "{build.path}/{build.project_name}.bin" -upload /sketch `

My programmers.txt

`## arduinoOTA as programmer. add entries with {ip} into programmers.txt
arduinoOTA.cmd={runtime.tools.arduinoOTA.path}/bin/arduinoOTA
arduinoOTA.program.params.verbose=
arduinoOTA.program.params.quiet=
arduinoOTA.ip=192.168.0.225
arduinoOTA.name=Arduino OTA (192.168.0.225)
arduinoOTA.program.tool=arduinoOTA
arduinoOTA.program.pattern="{cmd}" -address {ip} -port 65280 -t 30 -username "Arduino" -password "password" -sketch "{build.path}/{build.project_name}.bin" -upload /sketch -b`

alexsandro-ss, 2022-04-08:
Hi,

 My problem was in the construction of my platform.local . The example in the my_boards/avr folder worked for me.

My platform.local.txt looked like this

```
tools.arduinoOTA.cmd={runtime.hardware.path}/tools/arduinoOTA
tools.arduinoOTA.program.params.verbose=
tools.arduinoOTA.program.params.quiet=
tools.arduinoOTA.program.pattern="{cmd}" -address {ip} -port 65280 -t 30 -username arduino -password password -sketch "{build.path}/{build.project_name}.bin" -upload /sketch -b

```
Now everything is working. Thanks !