JAndrassy/ArduinoOTA #158: Add support for STM32F4xx

wowtah, 2022-06-01:
First of all, awesome job on adding STM32 support to the library.

Secondly, my request: 
Can you add **STM32F4** support also? I have tried to adapt the code to STM32F4 but I get some errors (mostly from stm32_flash_boot.c) that are above my knowledge level of the differences in memory structure between STM32F1 and F4 (PAGES vs SECTORS, etc).

JAndrassy, 2022-06-02:
I don't have the hardware. Once I had hope that the STM32 Arduino team would add STM32 support to ArduinoOTA library.

JAndrassy, 2022-06-02:
coffee inspires :-) I order a Black Pill. anyway it is the next most used STM32 Arduino

wowtah, 2022-06-02:
> coffee inspires :-) I order a Black Pill. anyway it is the next most used STM32 Arduino

Awesome :D

JAndrassy, 2022-06-11:
I started working on it today.

JAndrassy, 2022-06-13:
it's working.
now I have to consolidate the code. test again. then retest with the BluePill because of shared code.

JAndrassy, 2022-06-18:
you can try it

wowtah, 2022-06-19:
Awesome. I will check it out!

closed 2022-07-15 by JAndrassy

wowtah, 2022-08-11:
Works like a charm. Thanks so much!