JAndrassy/ArduinoOTA #163: Status Code 200 Received but New Program Not Loaded

JEG8, 2022-07-14:
Hi,

I am running into an issue on my Arduino IOT 33 where the OTA downloads the file and I receive a 200 status code, but the file is not replaced on the device. The old file is loaded back on. Do you know why this is happening? Could it be check sum related? The performance is very inconsistent.

Joe

JAndrassy, 2022-07-14:
there is no checksum. 
how do you test if the new or old version is running?

JEG8, 2022-07-14:
I have a firmware version number that gets updated when I add a new version to my server. Also, I can see that when I make edits to print statements, they do not show up on the serial monitor, indicating that the new version was never loaded on. However, I can see that the download occurs correctly because the size of the file matches the BIN output of the IDE.

JAndrassy, 2022-07-15:
you use InternalStorage directly?

JEG8, 2022-07-15:
No, I don't know what you mean by "InternalStorage". I am saving the downloaded file to a SPI Flash. I am then checking the size of the file, and comparing it to the "length" returned by the server. If they match, I proceed with the update. That all works fine. However, it appears that the bootloader portion isnt working properly. The file is not replaced on the hardware.

JAndrassy, 2022-07-15:
so you don't use the ArduinoOTA library at all.
you download the file to external SPI flash and it should be bootloaded by what?  I don't know about a library similar to SDU library, which would load the update binary from SPI flash as the SDU library does from SD card

JAndrassy, 2022-07-15:
SDU library is [just an array](https://github.com/arduino/ArduinoCore-samd/blob/master/libraries/SDU/src/boot/mkrwifi1010.h) set to be prepended by the linker before your sketch. And that array is a compiled [extra sketch](https://github.com/arduino/ArduinoCore-samd/blob/master/libraries/SDU/extras/SDUBoot/SDUBoot.ino) which loads the binary from SD card, writes it to flash and hands over the execution to regular sketch. It knows nothing about your SPI flash or the  Adaruit_SPIFlash libray.

JAndrassy, 2022-07-15:
> That file is actually on my SPI Flash, stored the same way. I should just need to point the SDU to look at my SPI Flash for the file, right?

right. you have to write and test the modified 'extras' sketch which will use the SPFlash library instead of the SD library. Then you have to convert the compiled binary of that sketch to an array of bytes in a .h file (there is some tool I found long time ago on Internet which can help with it). Then you create a library with that .h file and use it with your sketch.