JAndrassy/ArduinoOTA #168: ESP32 OTEthernet compilation error.

Arda-Bildik, 2022-09-07:
Hi,

I'm currently working on a project and I wanted to add OTA functionality with W5100.
I got the error "Compilation error for ESP32 Dev Module" while trying to upload the OTEthernet example.
This is the output from Arduino IDE:
[image]
I had a look in the code and tried including InternalStorage.h in the example code and 2 extra errors added to previous errors:
[image]

JAndrassy, 2022-09-08:
it is this problem
https://github.com/arduino-libraries/Ethernet/issues/193

closed 2022-09-08 by JAndrassy

Arda-Bildik, 2022-09-08:
> it is this problem [arduino-libraries/Ethernet#193](https://github.com/arduino-libraries/Ethernet/issues/193)

It seems like it I will try it. I'll post the fix if I find.

JAndrassy, 2022-09-08:
> > it is this problem [arduino-libraries/Ethernet#193](https://github.com/arduino-libraries/Ethernet/issues/193)
> 
> It seems like it I will try it. I'll post the fix if I find.

there is my comment