JAndrassy/ArduinoOTA #175: what if i reset?

KJPOUNDY132, 2022-10-26:
what if i rest after update (https://github.com/JAndrassy/ArduinoOTA/blob/master/examples/OTEthernet/OTEthernet.ino)
keep updated firmware or download one more time on every update

JAndrassy, 2022-10-27:
the library resets the MCU after applying the update. do you have some issue?

KJPOUNDY132, 2022-10-27:
i mean if i dont call update function on start up. it runs nothing or use last updated  firmware?

JAndrassy, 2022-10-27:
there are never two sketches at once. an Arduino sketch is build for exact addresses in flash. it can only run if it is at required address in flash

KJPOUNDY132, 2022-10-28:
i try to make ota trigger with mqtt.
if message equals OTA_OK esp32 gets new firmware.
but  the message is not equal OTA_OK use old firmware. is it posible?

JAndrassy, 2022-10-28:
why do you use this library for esp32?

KJPOUNDY132, 2022-10-28:
I haven't started trying because my Ethernet module hasn't arrived yet.
i just want to know;
Does the device download the file from the beginning every time it runs, or does it run the last file it received unless there is a new file?

JAndrassy, 2022-10-28:
it is your running sketch which downloads and applies the new version. it is on you how your sketch determines if there is a new version to download

closed 2022-11-06 by JAndrassy