JAndrassy/ArduinoOTA #178: [ESP32 on PlatformIO] error: conflicting declaration of C function 'sint8 close(SOCKET)' and 'class WiFiUDP' after Importing ArduinoOTA after WiFi.h

TNeutron, 2022-11-30:
I am getting the following error while trying to implement ArduinoOTA in my existing project that uses MQTT, NFC/RFID, TFT display, and a few buttons. 

```
Dependency Graph
|-- TFT_eSPI @ 2.4.61
|   |-- SPI @ 2.0.0
|   |-- SPIFFS @ 2.0.0
|   |   |-- FS @ 2.0.0
|   |-- FS @ 2.0.0
|   |-- LittleFS @ 2.0.0
|   |   |-- FS @ 2.0.0
|-- ESP32Time @ 1.0.4
|-- Adafruit MQTT Library @ 2.4.2
|-- MFRC522 @ 1.4.10
|   |-- SPI @ 2.0.0
|-- ArduinoOTA @ 2.0.0
|   |-- Update @ 2.0.0
|   |-- WiFi @ 2.0.0
|   |-- ESPmDNS @ 2.0.0
|   |   |-- WiFi @ 2.0.0
|   |-- WiFi101 @ 0.16.1
|   |   |-- SPI @ 2.0.0
|-- ESPmDNS @ 2.0.0
|   |-- WiFi @ 2.0.0
|-- SPI @ 2.0.0
|-- WiFi @ 2.0.0
|-- WiFi101 @ 0.16.1
|   |-- SPI @ 2.0.0

Building in release mode
Compiling .pio\build\esp32dev\src\main.cpp.o
In file included from .pio/libdeps/esp32dev/WiFi101/src/WiFiUdp.h:24,
                 from src/main.cpp:7:
.pio/libdeps/esp32dev/WiFi101/src/socket/include/socket.h:1839:15: error: conflicting declaration of C function 'sint8 close(SOCKET)'
 NMI_API sint8 close(SOCKET sock);
               ^~~~~
In file included from C:/Users/tnowr/.platformio/packages/framework-arduinoespressif32/tools/sdk/esp32/include/newlib/platform_include/sys/unistd.h:23,
                 from c:\users\tnowr\.platformio\packages\toolchain-xtensa-esp32\xtensa-esp32-elf\sys-include\unistd.h:4,   
                 from c:\users\tnowr\.platformio\packages\toolchain-xtensa-esp32\xtensa-esp32-elf\sys-include\pthread.h:25, 
                 from C:/Users/tnowr/.platformio/packages/framework-arduinoespressif32/tools/sdk/esp32/include/newlib/platform_include/pthread.h:21,
                 from c:\users\tnowr\.platformio\packages\toolchain-xtensa-esp32\xtensa-esp32-elf\include\c++\8.4.0\xtensa-esp32-elf\no-rtti\bits\gthr-default.h:48,
                 from c:\users\tnowr\.platformio\packages\toolchain-xtensa-esp32\xtensa-esp32-elf\include\c++\8.4.0\xtensa-esp32-elf\no-rtti\bits\gthr.h:151,
                 from c:\users\tnowr\.platformio\packages\toolchain-xtensa-esp32\xtensa-esp32-elf\include\c++\8.4.0\ext\atomicity.h:35,
                 from c:\users\tnowr\.platformio\packages\toolchain-xtensa-esp32\xtensa-esp32-elf\include\c++\8.4.0\bits\basic_string.h:39,
                 from c:\users\tnowr\.platformio\packages\toolchain-xtensa-esp32\xtensa-esp32-elf\include\c++\8.4.0\string:52,
                 from c:\users\tnowr\.platformio\packages\toolchain-xtensa-esp32\xtensa-esp32-elf\include\c++\8.4.0\stdexcept:39,
                 from c:\users\tnowr\.platformio\packages\toolchain-xtensa-esp32\xtensa-esp32-elf\include\c++\8.4.0\array:39,
                 from c:\users\tnowr\.platformio\packages\toolchain-xtensa-esp32\xtensa-esp32-elf\include\c++\8.4.0\tuple:39,
                 from c:\users\tnowr\.platformio\packages\toolchain-xtensa-esp32\xtensa-esp32-elf\include\c++\8.4.0\functional:54,
                 from C:/Users/tnowr/.platformio/packages/framework-arduinoespressif32/cores/esp32/HardwareSerial.h:49,     
                 from C:/Users/tnowr/.platformio/packages/framework-arduinoespressif32/cores/esp32/Arduino.h:174,
                 from src/main.cpp:4:
c:\users\tnowr\.platformio\packages\toolchain-xtensa-esp32\xtensa-esp32-elf\sys-include\sys\unistd.h:30:9: note: previous declaration 'int close(int)'
 int     close (int __fildes);
         ^~~~~
In file included from src/main.cpp:7:
.pio/libdeps/esp32dev/WiFi101/src/WiFiUdp.h:35:7: error: redefinition of 'class WiFiUDP'
 class WiFiUDP : public UDP {
       ^~~~~~~
In file included from C:/Users/tnowr/.platformio/packages/framework-arduinoespressif32/libraries/WiFi/src/WiFi.h:39,        
                 from src/main.cpp:5:
C:/Users/tnowr/.platformio/packages/framework-arduinoespressif32/libraries/WiFi/src/WiFiUdp.h:42:7: note: previous definition of 'class WiFiUDP'
 class WiFiUDP : public UDP {
       ^~~~~~~
Compiling .pio\build\esp32dev\lib852\ESPmDNS\ESPmDNS.cpp.o
Compiling .pio\build\esp32dev\libdac\WiFi101\WiFi.cpp.o
Compiling .pio\build\esp32dev\libdac\WiFi101\WiFiClient.cpp.o
*** [.pio\build\esp32dev\src\main.cpp.o] Error 1
In file included from C:/Users/tnowr/.platformio/packages/framework-arduinoespressif32/tools/sdk/esp32/include/newlib/platform_include/sys/unistd.h:23,
                 from c:\users\tnowr\.platformio\packages\toolchain-xtensa-esp32\xtensa-esp32-elf\sys-include\unistd.h:4,
[... 54 lines cut ...]
*** [.pio\build\esp32dev\libdac\WiFi101\WiFi.cpp.o] Error 1
```

I have the following order of library imports.

```
#include <Arduino.h>
#include <WiFi.h>

#include <ESPmDNS.h>
#include <WiFiUdp.h>
#include <ArduinoOTA.h>

#include <pgmspace.h>
#include <TFT_eSPI.h>
#include <MFRC522.h>
#include <SPI.h>
#include <ESP32Time.h>
#include "Adafruit_MQTT.h"
#include "Adafruit_MQTT_Client.h"
```

I am consciously removing WiFi101 library from PlatformIO project library list but however, but it keeps coming back again and again after each complication. My assumption is that ArduinoOTA has a dependency on WiFi101 and that results in this conflict, with the Wifi.h Library. 

How to solve this issue?

JAndrassy, 2022-11-30:
don't use this library with esp32 WiFi

closed 2022-11-30 by JAndrassy

TNeutron, 2022-11-30:
> don't use this library with esp32 WiFi

The code examples in the Arduino library also use the WiFi library like 
```
#include <WiFi.h>
#include <ESPmDNS.h>
#include <WiFiUdp.h>
#include <ArduinoOTA.h>
``` 

But using this same pattern. I have found that working but the same pattern in my other project (in PlatformIO) seems to be causing this issue. Any thoughts on that?

JAndrassy, 2022-11-30:
use the ArduinoOTA library bundled with the esp32 platform