JAndrassy/ArduinoOTA #182: STM32F103C Blue Pill "Internal Server Error"

aaroncake, 2022-12-31:
Hello!

I'm attempting to use the library with an STM32F103C generic "Blue Pill" board. Using the latest stm32duino core. 

Looks like the sketch uploads, but then after flashing the error returned is "Internal Server Error".

Using the example "OTEthernet" sketch included with the library so assuming internal storage.

Anything I need to do to get this working? Specific compile options? A specific bootloader?

Thanks!

JAndrassy, 2022-12-31:
I guess you have F103C**8** selected in Tools menu. 64 kB flash is not enough. you need F103C**B** with 128 KB flash. check which one is on your BluePill

aaroncake, 2022-12-31:
Oh for the love of... :)

I knew it had to be something simple. I didn't even think that the option in the Tools menu would limit flash use to 64K, and assumed my C8 had only 64K. But I learned something today after my Googling that C8s are actually often CBs and almost certainly have 128K. 

I'll try it tomorrow morning and report back.

Thanks!

aaroncake, 2023-01-01:
Well, looks like my BluePills are 64K.

Tried compiling with the "(or C8 with 128K)" option and received the same "Internal Server Error" on upload.

Added a check to my code to print out flash size:

const size_t FLASH_SIZE = (*((uint16_t*)FLASH_SIZE_DATA_REGISTER)) << 10;

Serial.println(FLASH_SIZE);

And my debug output was:

65536
Initialize Ethernet with DHCP:
  DHCP assigned IP 1.1.1.15

(yes my IP range is 1.1.1.0)

Now I'm off to Aliexpress to try to find some 128K Blue Pills.

Thank you for your help!

aaroncake, 2023-01-02:
I can't find any F103C8 boards which "guarantee" 128K, and no CB boards seem to be available.

So I'm just going to order to Black Pill F411 boards as I see they are on your tested list.

(I'm making a project that needs a fair bit of I/O and I want others to be able to make it as well, otherwise I'd just use an Atmega 1284 on a custom board)

JAndrassy, 2023-01-03:
> (I'm making a project that needs a fair bit of I/O and I want others to be able to make it as well, otherwise I'd just use an Atmega 1284 on a custom board)

Arduino Mega?
MKR Zero with MKR ETH shield?

aaroncake, 2023-01-04:
Unfortunately no physical space for the Mega, and to be honest, I had just not thought of the MKR line. I don't have any and had intended to make this project without purchasing any further parts. Which sort of went out of the window having to buy the Black Pills.

aaroncake, 2023-03-04:
Success!

My STMF4 "Black Pill" boards arrived a while ago and I finally got around to ArduinoOTA just a few minutes ago.

Worked first try.

Uploading 'Network_PDU' to 'Generic STM32F4 series                                                                                                      (STMicroelectronics_GenF4)' using 'STM32F4 OTA (1.1.1.17)'
Uploader started for board Generic STM32F4 series                                                                                                      (STMicroelectronics_GenF4)
Uploader will use programmer name: arduinoOTA14
C:\Users\Administrator\AppData\Local\arduino15\packages\arduino\tools\arduinoOTA\1.3.0\bin\arduinoOTA -v -address 1.1.1.17 -port 65280 -username arduino -password password -sketch "C:\Users\Administrator\AppData\Local\Temp\VMBuilds\Network_PDU\STMicroelectronics_GenF4\Debug/Network_PDU.ino.bin" -upload /sketch -b
Connecting to board ...  done
Uploading sketch ...  done
Flashing sketch ...  done
	The upload process has finished.
OK
Sketch uploaded successfully

Thank you for all your help!

If it helps, I confirm that an F411CE board works with Wiznet W5500 with the code compiled in Visual Studio using Visual Micro, hardware programmer as defined in programmers.txt using your example.

The "Buy Me A Coffee" link seems to be removed from readme.md with your last commit in Feb. Is it still valid?

JAndrassy, 2023-03-04:
> The "Buy Me A Coffee" link seems to be removed from readme.md with your last commit in Feb. Is it still valid?

it is now mentioned here https://github.com/JAndrassy

closed 2023-03-04 by JAndrassy