JAndrassy/ArduinoOTA #183: STM32F103RB Nucleo 64 ...No OTA programmer option in IDE

fostiras, 2023-01-23:
Hello !
 I have successfully compiled and uploaded the OTEthernet example (InternalStorage) and also got the Network Port option whenever the shield is connected to my Lan and gets an IP.  
But I can't get an OTA option in the Upload method , when trying to upload a compiled firmware with the network port selected, the IDE still tries to find the STM32CubeProgrammer. 
If the Nucleo is connected to a USB from the PC it does the upload through the COM port (COM3 in my case) although the network port is selected and not COM3.
If the Nucleo is connected form a USB charger -  for power only - it will not upload searching for the missing STM32CubeProgrammer.
I tried to copy the lines from the platform.local.text (in /extras)/stm32/) into the platform.txt ( in \packages\STMicroelectronics\hardware\stm32\2.4.0 ) with no luck . The programmers.txt file in the same location is empty and I don't know what to put in it.
My hardware is a Nucleo 64 STM32F103RB board with a W5100 ethernet shield on it.
I am using Arduino portable 1.8.16 with STM32 core 2.4.0 .
Thank you !

fostiras, 2023-01-23:
Update, That was quick !!
I copied these lines in programmers.txt   (in /extras)/stm32/) into the platform.txt ( in \packages\STMicroelectronics\hardware\stm32\2.4.0 ) :

arduinoOTA.name=Arduino OTA (192.168.1.0)
arduinoOTA.program.tool=arduinoOTA
arduinoOTA.program.default=arduinoOTA
arduinoOTA.ip=192.168.1.0

and all went well !!!
No matter what IP I put in programmers.txt it will upload and flash the board - of course the network port shows the correct one (192.168.1.20 in my case.
I will try the same with two same boards on the Lan to see if I can switch between the too only through the network port.

JAndrassy, 2023-01-24:
the programmer definition is not necessary for network port. it just sometimes takes time to be detected

closed 2023-01-24 by JAndrassy

fostiras, 2023-01-24:
As mentioned the network port with the correct mDNS name and the correct IP is displayed immediatly upon starting the arduino IDE. But without the definitions I was unable to upload the firmware.