JAndrassy/ArduinoOTA #184: RP2040 connect with Arduino-pico core hangs

AcThPaU, 2023-01-30:
OTA does not work on RP2040 connect with [Arduino-pico](https://github.com/earlephilhower/arduino-pico) core. The ArduinoOTA lib was porpely used and the OTA came from the Arduino-pico core was deleted. 

The MCU hangs when calling ``WiFi.localIP()``, otherwise ``WiFiNINA.h`` works fine, even NTP client works.

Tested on WIFININA firmware 1.4.8 and 1.5.0

vshymanskyy, 2023-03-09:
I can confirm the issue with WiFiNINA `localIP()`, but this has nothing to do with ArduinoOTA library.
There's an issue in Arduino-pico core probably.

vshymanskyy, 2023-03-09:
https://github.com/earlephilhower/arduino-pico/issues/973 ?

AcThPaU, 2023-03-09:
my A4 and A5 are used as I2C, it still hangs

vshymanskyy, 2023-03-12:
Yeagh, same issue here. However I think it needs to be reported Arduino-pico core, and this issue is not related to  ArduinoOTA

closed 2023-03-12 by JAndrassy