JAndrassy/ArduinoOTA #198: After update only espota tool

jhnwmr, 2023-04-15:
Hello,
I have updated the ESP32 core from 1.6.0 to 2.0.7 and IDE itself to 2.0.3. Now I can not upload OTA updates.
I replaced the original ArduinoOTA library with this one and added the platform.local.txt.
Compiling works fine, but uploading does not. It always uses the espota tool. 
However, the ArduinoOTA tool should be used.
What do I have to change so that the flashing works correctly?
Thanks!

JAndrassy, 2023-04-16:
try this platform.local.txt for IDE 2
[platform.local.txt](https://github.com/JAndrassy/ArduinoOTA/files/11242496/platform.local.txt)

lcoffin, 2023-04-16:
Cool. I’ll give that a try when I’m able to. I’m traveling for the week, so won’t be until I’m back.-L

closed 2023-04-18 by jhnwmr