JAndrassy/ArduinoOTA #199: SD Issues with SD.h lib

CptHolzschnauz, 2023-04-17:
In examples/Advanced/OTASketchDownload_SD)/OTASketchDownload_SD.ino
It's:
`#define SDCARD_SS_PIN 4`
And then 
`if (!SD.begin(SDCARD_SS_PIN)) {`
And then
`File file = SD.open(BIN_FILENAME, O_CREAT | O_WRITE);`

Which leads in my project to a SD failure trying SD.open .

With 
`if (!SD.begin(4)) {`
and
` File file = SD.open(BIN_FILENAME, FILE_WRITE);`

It worked. The SD.begin(4) is probably superstitious because the compiler should come with the same code, the FILE_WRITE is maybe wanted by a newer SD.h version. The pure example as it is works, but as i said not with a longer sketch. Don't know why.

Suggestion: Replace 
`File file = SD.open(BIN_FILENAME, O_CREAT | O_WRITE);`
with
` File file = SD.open(BIN_FILENAME, FILE_WRITE);`

CptHolzschnauz, 2023-04-17:
mkr1010

JAndrassy, 2023-04-18:
what error do you get?

CptHolzschnauz, 2023-04-18:
`if (!SD.begin(SDCARD_SS_PIN)) {`
Can't initialize (So return 0)
`File file = SD.open(BIN_FILENAME, O_CREAT | O_WRITE);`
Can't open/create file. 
Sry, i'm in transit away from my dev desk so i can't give more precise answer at the moment.
It's not the card or it's filesystem, i checked that.

CptHolzschnauz, 2023-04-19:
Yes. Your comment makes sense, the 
`#define FILE_WRITE (O_READ | O_WRITE | O_CREAT | O_APPEND)`
is a bit strange bacause it means everything, like a general approach but thats a SD thing.

` File file = SD.open(BIN_FILENAME, O_CREAT | O_WRITE);` makes sense but this may cause issues.
In the SD lib examples they check for a existing file, if existing remove it and then make a new one. Maybe this approach is more failsafe in SD.h, i don't know. I will do some further testing when i'm back and report here. Thx & you rock!

CptHolzschnauz, 2023-07-13:
Just a thought: Why not use sdfat.h instead of sd.h? sd.h is a old fork of sdfat.h. sdfat.h seems to be more maintained.

JAndrassy, 2023-07-21:
I like the SD library

closed 2023-07-21 by JAndrassy