JAndrassy/ArduinoOTA #200: OTA upload succesfull, but old sketch remains, even when I JUST flashed the Optiboot bootloader

ChadMatsalla, 2023-04-20:
### Discussed in https://github.com/JAndrassy/ArduinoOTA/discussions/151

<div type='discussions-op-text'>

<sup>Originally posted by **AttilaTtila** April 24, 2022</sup>
Hi Juraj,

I have 2x Mega2560 (with builtin Wifi) boards and i'd really like to make OTA working for them!
So i burned the Optiboot bootloader by MCUDude.
Then i used your example sketch (SerialWiFiOTA) to test. Then i make minor change in the code and upload it via "Upload using programmer". All seems OK, "Sketch uploaded successfully" message appears, but after board resets, the old sketch started running again (or it just hangs completely).
Any advise please?
Thanks!
Attila
</div>

JAndrassy, 2023-04-20:
If you test OTA with the 'fake' programmer, close the Serial Monitor in IDE. With Serial Monitor open, IDE will reset the board after upload tool finishes. The board then doesn't finish to apply the update.

ChadMatsalla, 2023-04-20:
What about when arduinoOTA is used? Definitely no serial connection is open to the board.

JAndrassy, 2023-04-20:
if you use network port then no, Serial Monitor is not connected. but with ota programmer method the Serial Monitor could be connected.

are you on Windows or on Linux? I had problems on Linux with upload command in quotes.

insufficient powering could be a problem. doing wifi and writing to flash at the same time.

Bixilon, 2026-10-12:
@JAndrassy I am experiencing something similar with an esp8266. "Sketch uploaded successfully" and the board is reset, but the old firmware is starting. What can I do to track this down?

Thanks!