JAndrassy/ArduinoOTA #203: Upload returns OK but the sketch is not replaced STM32 Nucleo

Yaroki2, 2023-05-08:
Hello!
I do OTA use NUCLEO-F303RE and get
[image]
Then  MCU don't resets and stuck forever
i tried closing the Serial Monitor in the IDE but it doesn't work

JAndrassy, 2023-05-08:
sorry, I don't have an STM32 F3 so it was never tested. The F3 has the same flash operations as F1 and I tested F1 so there was a chance it could work.

Yaroki2, 2023-05-08:
Thanks you for the reply!
I realize I need to press the manual reset button to get it to work, do you have any ideas?

JAndrassy, 2023-05-08:
is the sketch updated?

Yaroki2, 2023-05-08:
Yes, Sketch updated but doesn't reset automatically

JAndrassy, 2023-05-08:
InternalStorage.apply() calls at the end `NVIC_SystemReset();` which is a reset command for all ARM MCU and works on STM32 too.
you could test it in a simple sketch.

closed 2023-05-09 by Yaroki2

Yaroki2, 2023-05-09:
I tested `NVIC_SystemReset();` with simple sketch and it works fine.
I think `NVIC_SystemReset();` is the reason, when i change other way to reset in library it works successfully.