JAndrassy/ArduinoOTA #206: MKR1310 with ETHERNET Port

alexair777, 2023-05-25:
Hi,

I am using an Arduino MKR1310WAN together with Arduino MKR ETH Shield.
It is working until the password.
When I am typing in the password I get the following message:
"Failed uploading: uploading error: signal: killed"

What is wrong with this?
Thanks a lot.

JAndrassy, 2023-05-26:
https://forum.arduino.cc/t/arduino-nano-iot-33-arduinoota-not-uploading/1072889/20

alexair777, 2023-06-13:
Hi,

thanks a lot for your reply :)
I tried this recommendation of the forum but without any success:

"I figured out what the issue was, there's a problem with the ArduinoOTA install and replacing the one used by the IDE by one recompiled from the github repo fixed whatever it was." -> i downloaded the .zip repo and put it in the arduino library path.

I can select the IP port from the microcontroller MKR WAN 1310 with MKR Ethernet LAN Adapter but when I type in the standard pwd with password the following message occurs while uploading a new sketch.

Best regards

alexair777, 2023-06-14:
Hi,

now I figured out that the OTA upload works on a windows machine without any problems, but on mac not...
Do you now, what this problem could be?
Should the serial monitor be available with IP port?

Thanks a lot!

Best regards

JAndrassy, 2023-06-14:
the forum thread explains that it is a wrong binary for the upload tool (arduinoOTA) in the IDE for Mac and you have to replace it

JAndrassy, 2023-06-16:
sorry I thought there was a solution described in that thread.

https://github.com/arduino/arduinoOTA/releases

you can see the location of the arduinoOTA in IDE console of you turn on verbose mode in IDE Preferences

replace it

alexair777, 2023-06-16:
wow, thats amazing, its working now! thanks a lot for your great help! :)

closed 2023-06-17 by JAndrassy