JAndrassy/ArduinoOTA #207: Support for Renesas MCUs

vshymanskyy, 2023-07-08:
As Arduino has released [`Portenta C33`](https://store.arduino.cc/products/portenta-c33) and [`UNO R4`](https://store.arduino.cc/pages/unor4), please consider adding these boards.

JAndrassy, 2023-07-08:
yes I want to add at least the Uno R4. but it is not available in Europe yet. only as pre-order in some e-shops.
Arduino offered samples for developers some months ago, but I didn't think it would be interesting.

mikeycoding, 2023-07-31:
Hi JAndrassy,

the Uno R4 has been released and I already own one. The OTA support is heavily requested in the Arduino forums as well on other platforms.

Is there any ETA on when you will release support for the R4 WiFi? Thank you.

vshymanskyy, 2023-07-31:
There is no active work being done on it, so why asking about ETA?

mikeycoding, 2023-07-31:
> There is no active work being done on it, so why asking about ETA?

Why so rude? I just wanted to kindly ask if the author has any plans to implement it and if so, whether there is an estimation on when it will be done.

JAndrassy, 2023-07-31:
so I am working on it. the Renesas SDK library bundled with the core is pre-build by Arduino without flash writing functions. so my next step is to install the Renesas toolchain and rebuild the lib.
also the bundled BlockDevice Arduino library  works only for Portenta C33 'code flash' write, but that is easy to fix.
If Arduino works on Cloud OTA, they will have to make this steps too.

JAndrassy, 2023-08-23:
done

closed 2023-08-23 by JAndrassy

yann-l, 2023-10-02:
Is there any change to have it ported for Portenta C33 ?

JAndrassy, 2023-10-03:
For C33 you can use the SFU library. The UpdateFromFile example show how to store the update binary for the SFU library.

If you want to upload from IDE with this library, you can use custom Storage as demonstrated by the CustomStorageNanoConnect example.

yann-l, 2023-10-03:
Ho excellent!
Indeed SFU is easy to use, but the download part were missing, and downloading a big file like an ota file with SSLClient is not possible (https://stackoverflow.com/questions/75941895/why-does-my-https-request-get-disconnected-on-large-files). My attempts with a 2-3ko file failed (over SSL only). an ota file is at lease 300ko.

Reading the new PR, they use SSLClient to download the file, so I'm a bit septic, but I'll give it a try soon!

JAndrassy, 2023-10-04:
@yann-l,  you want download from a server or upload from the IDE?

yann-l, 2023-10-04:
I want to download from a server. https://github.com/arduino/ArduinoCore-renesas/pull/157 is perfect, but it needs https://github.com/arduino/ArduinoCore-renesas/pull/134 to work correctly. With both applied, I was able to make it work.