JAndrassy/ArduinoOTA #209: Compilation fails for end() method

bakadave, 2023-07-20:
I'm trying to handle ethernet disconnection scenarios, however if I try to call `ArduinoOTA.end()` compiler returns with error:
`class EthernetServer' has no member named 'stop'`

Upon closer inspection it looks like EthernetServer does not have a `stop()` method afterall. 
What is the recommended way to proceed? What happens if I never call `end()`? Should I call `begin()` after reconnection?

JAndrassy, 2023-07-20:
I don't know what happens when server.begin() of Ethernet library is called again. it will try to create a new server with the same port and I don't know if it ends with error or if it creates it and how will it then behave.

bakadave, 2023-07-20:
Thank you. I'm not familiar enough with these libs to figure it out on my own but I'm happy to help if you give me some pointers.

bakadave, 2023-07-21:
**Workaround until the answer is found:** in case of ethernet disconnect I restart the microcontroller and wait for a normal ethernet connection to be estabilished. Once ethernet is (re)estabilished `ArduinoOTA.begin()` is called (only once).
This avoids the undefined behaviour detailed above.