JAndrassy/ArduinoOTA #217: Renesas get stuck in `InternalStorage.open()`

vshymanskyy, 2023-08-28:
`debugPrint()`:
```log
storageStartAddress 22000
SKETCH_START_ADDRESS 4000
MAX_FLASH 40000
```

I call `InternalStorage.open(56296)` directly.
AFAIK, the call freezes inside https://github.com/JAndrassy/ArduinoOTA/blob/f88941c166e7c5ca475f3393565ac553879df586/src/utility/r_flash_lp_cf.c#L424

vshymanskyy, 2023-08-28:
@JAndrassy please take a look

JAndrassy, 2023-08-28:
I tested
```
#include <ArduinoOTA.h>
void setup() {
  Serial.begin(115200);
  Serial.println("TEST START");
  InternalStorage.debugPrint();
  InternalStorage.open(56296);
  InternalStorage.close();
  Serial.println("TEST OK");
}
void loop() {
}
```
and the output is
```
TEST START
storageStartAddress 22000
SKETCH_START_ADDRESS 4000
MAX_FLASH 40000
TEST OK
```
so I can't reproduce the problem.

vshymanskyy, 2023-08-28:
Thanks! Looks like I crashed my board and it was completely unrelated to ArduinoOTA.
I can confirm that OTA works:
```
[16234] OTA chunk @  c400, size: 1024, crc: 3cf30dc8 => OK
[16302] OTA chunk @  c800, size: 1024, crc: 4b045b5f => OK
[16369] OTA chunk @  cc00, size: 1024, crc: 9cd3b7fc => OK
[16438] OTA chunk @  d000, size: 1024, crc: d7357ffb => OK
[16507] OTA chunk @  d400, size: 1024, crc: 5e9a4d25 => OK
[17572] OTA chunk @  d800, size: 1000, crc: dde66a84 => OK
[18597] CRC32 verified: db6d24fd
[18600] Applying the update

[104] Main firmware: 0.2.0
[107] Build: Aug 28 2023 17:35:43
```

closed 2023-08-28 by vshymanskyy