JAndrassy/ArduinoOTA #219: Unable to compile Arduino Code with ArduinoOTA.begin()

mskressin, 2023-08-29:
I am sorry if this is a dumb mistake on my part, but I cannot figure out what I am doing wrong. When I try the compile using IDE 2.2.0 the following code I am getting the error: "Compilation error: 'ArduinoOTA' was not declared in this scope" I have installed via the library manager ArduinoOTA version 1.0.10 and the board package for my Uno R4 WIFI board (version 1.02). I also followed the install instructions and copied the file platform.local.txt from the extras/renesas folder to the board install folder renesas_uno/1.02. 

**Simple code I am trying to compile:**
#include <ArduinoOTA.h>
#include <WiFiS3.h>        

const char* ssid = "MyRouter";  // Change to your WiFi Network name
const char* password = "MyPassword";

void setup() {
  Serial.begin(115200);
  WiFi.begin(ssid, password); 

  // Ensure WiFi is connected
  while (WiFi.status() != WL_CONNECTED) {
    delay(500);
  }
  ArduinoOTA.begin();
}

void loop() {
  ArduinoOTA.handle();  // Handles a code update request
  delay(1000);
}

Compiling this gives me the following error:
In function 'void setup()':
error: 'ArduinoOTA' was not declared in this scope
   ArduinoOTA.begin();
   ^~~~~~~~~~
note: suggested alternative: 'ArduinoOTAClass'

Using library ArduinoOTA at version 1.0.10 in folder: Documents/Arduino/libraries/ArduinoOTA 
Using library WiFiS3 at version 0.0.0 in folder: Library/Arduino15/packages/arduino/hardware/renesas_uno/1.0.2/libraries/WiFiS3 
exit status 1

Compilation error: 'ArduinoOTA' was not declared in this scope

JAndrassy, 2023-08-30:
Uno R4 support is in 1.0.11 which is not yet published to Library Manager

closed 2023-08-30 by JAndrassy

mskressin, 2023-08-30:
I uninstalled 1.0.10 and installed 1.0.11 via the library manager and still have the same error. I am I doing something wrong?

JAndrassy, 2023-08-30:
open the WiFi101_OTA example and change the WiFiNINA include to WiFiS3. does that compile?

mskressin, 2023-08-30:
Thanks, that help me track down the problem. It seems the include order is critical. I had:
#include <ArduinoOTA.h>
#include <WiFiS3.h> 
But it needed:
#include <WiFiS3.h> 
#include <ArduinoOTA.h>
Has do do with the way that the code in ArduinoOTA is defining the class: 
#elif defined(WiFiNINA_h) || defined(WIFI_H)  || defined(WiFiS3_h) || defined(ESP8266) || defined(ESP32) || defined(ARDUINO_RASPBERRY_PI_PICO_W) // NINA, WiFi101 and Espressif WiFi
#ifdef NO_OTA_PORT
ArduinoOTAClass  <WiFiServer, WiFiClient> ArduinoOTA;
#else
#include <WiFiUdp.h>
ArduinoOTAMdnsClass <WiFiServer, WiFiClient, WiFiUDP> ArduinoOTA;
#endif 
Apparently, the pre-complier could not deal with the fact that variable WiFiS3_h was defined after the header file was included. Makes sense now that I found it.
Thanks again for your help.