JAndrassy/ArduinoOTA #220: Error compiling for board Arduino Mega 2560 (Optiboot).

youssiefkhaled, 2023-09-09:
` file included from C:\Users\Yousif\Documents\Arduino\libraries\WiFiNINA\src/WiFiStorage.h:23:0,
                 from C:\Users\Yousif\Documents\Arduino\libraries\WiFiNINA\src/WiFi.h:37,
                 from C:\Users\Yousif\Documents\Arduino\libraries\WiFiNINA\src/WiFiNINA.h:23,
                 from C:\Users\Yousif\Documents\Arduino\libraries\ArduinoOTA-master\examples\Advanced\OTASketchDownloadWifi\OTASketchDownloadWifi.ino:24:
C:\Users\Yousif\Documents\Arduino\libraries\WiFiNINA\src/utility/wifi_drv.h:293:12: error: 'PinStatus' does not name a type
     static PinStatus digitalRead(uint8_t pin);
`
i can't compile the arduinoOTA sketches ? 
for both 
WIFI101_OTA
OTAsketchDownloadWifi 
i also tested serialWIFIOTA it detects the esp but i dont know how to make it connect to the wifi network 
any help?

youssiefkhaled, 2023-09-09:
> 

ok the solution was simply to replace the wifininawith wifiespat lib

closed 2023-09-09 by youssiefkhaled

JAndrassy, 2023-09-09:
https://github.com/JAndrassy/WiFiEspAT#persistent-wifi-connection

youssiefkhaled, 2023-09-09:
just a qustion 
i should now ulpoad this SetupWiFiConnection in the esp-01 and join my network 
after that what i should expect ?
a network port ?
or i will have to use the fake programmer?
 @JAndrassy

youssiefkhaled, 2023-09-09:
i think i got it know i will come after testing

youssiefkhaled, 2023-09-09:
ok what i did is reuploaded the serial ota with network connected 
and 
[image]
uncommented to get the static ip 
now there is no network port nor and i cant upload throw the "fake programmer"
i also edited the fake programer to static ip but i get this 

'connecting to board ... failed!
Error flashing the sketch
Error flashing the sketch
'

youssiefkhaled, 2023-09-09:
i just had to move the platform.local
thank youuuuuuuuuuu