JAndrassy/ArduinoOTA #221: Password wrong

LukeScrewdriver, 2023-10-14:
Hi, I'm trying to upload code to an arduino mega and an arduino nano 33 iot.

**Attempting on ide 2.x (for 33 Nano iot):**
When I try to upload the code it does not ask me for the password
I enabled the detailed output and noticed that it uses 'arduino' as the password.
I tried to run the program it points me to from the console with the correct parameters and it loaded correctly.

**Attempted on ide 1.x (Arduino Mega):**
When I try to load the code it asks me for the password
I enter the password (which I verified) and then it asks for it over and over again as if it were wrong.
I enabled the detailed output and noticed that the password parameter is not entered.

JAndrassy, 2023-10-14:
password is the third parameter in ` ArduinoOTA.begin(WiFi.localIP(), "Arduino", "password", InternalStorage);`

for Mega you didn't copy platform.local.txt from extras to the boards package. it now uses the upload command for ArduinoYun.
https://github.com/JAndrassy/ArduinoOTA#atmega-support

LukeScrewdriver, 2023-10-15:
I remembered installing them... now they are installed 
[image]

> password is the third parameter in  ArduinoOTA.begin(WiFi.localIP(), "Arduino", "password", InternalStorage);

Yes i changed the name and the password field

> do you have the latest version of the SAMD boards package?

Yes

JAndrassy, 2023-10-15:
so now the upload command for Mega is OK?

LukeScrewdriver, 2023-10-15:
Yes the upload to the mega has settled down.
I have another small problem but I will create another issue for that.
Thank you for responding to me quickly

closed 2023-10-15 by LukeScrewdriver