JAndrassy/ArduinoOTA #222: Compatibility with Ethernet3 library?

modmt, 2023-10-16:
Hello,

I tried to use the OTA functionality in combination with an Atmega 1284P and the ethernet3.h library (https://github.com/sstaub/Ethernet3) but I always get the error: "ArduinoOTA' was not declared in this scope
 ArduinoOTA.poll();".

Do you know if the ArduinoOTA library is compatible with the Ethernet3 library? I have to use this library because of its additional features which the Standard Ethernet library cannot provide.

Thank you for your help.
Matthias

JAndrassy, 2023-10-16:
why do you use the Ethernet3 library? the Arduino Ethernet library works good with W5500

modmt, 2023-10-16:
I use the "Ethernet.phyMode(FULL_DUPLEX_10);" to reduce the Power consumption. As far as I know the standard Ethernet library doesn't have this function.
Also choosing an custom RST and CS pin is more easy with the Eternet3 library.

JAndrassy, 2023-10-16:
add `#define ethernet_h` before `#include <ArduinoOTA.h>`

modmt, 2023-10-17:
Thank you very much for your quick response! 

After adding the #define ethernet_h compiling the programm works fine.

Now I have an additional problem: Connecting to the board and uploading the sketch seems to works fine (see attachment) but flashing does not work and I'm running out of ideas why.
[image]

In my sketch I have in addition to my code the following code snippets:

#define ethernet_h
#define NO_OTA_PORT
#include <ArduinoOTA.h>

void setup() {
  ArduinoOTA.begin(Ethernet.localIP(), "Arduino", "password", InternalStorage);
....
}

void loop() {
ArduinoOTA.poll();
...
}

I added the following to my programmer.txt in the MightyCore folder (IP 192.168.188.100 for the ATmega1284P board is static and all the functions (like MQTT) using the ethernet are working)

arduinoOTA100.name=Arduino OTA (192.168.188.100)
arduinoOTA100.program.tool=arduinoOTA
arduinoOTA100.program.tool.default=arduinoOTA
arduinoOTA100.ip=192.168.188.100

and also copied the platform.local.txt from the ArduinoOTA container to this folder.
After that I found the programmer in the menu.

I also burned the Optiboot bootloader to the Atmega1284P.

Do you know what else I can do to get OTA flashing working? Did i miss anything in the procedure?

Thank you in advance.
Matthias

JAndrassy, 2023-10-17:
a simple example like the OTEthernet works?
it could be a timeout in the upload tool for larger sketch.

modmt, 2023-10-17:
I tried it with the simple example with the same result. 
The only change I made is using the ethernet3 library instead of the ethernet library. The port was recongized in the arduino IDE and I have to type the password. After that uploading works, but flashing not.
The same with the fake programmer.

I think it's not a timeout because I got the error after 5-8 seconds, even with small sketches to check.

I also added this, but no change:
[image]

As I understood the Optiboot bootloader is already included in the MightyCore.
Is it enough to chose the bootloader and after that burn bootloader like this in the MightyCore settings or do I have to flash the optiboot (8) seperately? Maybe this could be an issue I made wrong.
[image]

JAndrassy, 2023-10-17:
turn on upload verbose mode in IDE Preferences