JAndrassy/ArduinoOTA #223: STM32F407 - Flash Size issue

gcharles81, 2023-10-23:
Hello I have been using this great library for OTA over Wifi  of my project using STM32F407VE mcu with 512kb of flash.

Everything was working fine for months but now as my code evolves the .bin file is getting large and 155kb in size and with this size the update will be halted with serial message "There is not enough space to store the update. Can't continue with update." 

After that I have updated the MCU from STM32F407VE 512KB to STM32F407VG 1Mb but the issue still happen , can it be the Internalstorage class is calculating the available space wrong for the STM32F407VG ? 

STM32F407VE compilation details - firmware uses of  Flash: [===       ]  30.2% (used 158208 bytes from 524288 bytes)
[image]
  

STM32F407VG compilation details - firmware uses of Flash: [==        ]  15.1% (used 158208 bytes from 1048576 bytes)
[image]

JAndrassy, 2023-10-24:
ArduinoOTA uses only the first 'bank' of the flash memory on STM32. and max bank size is 512 kB. 
```
        SKETCH_START_ADDRESS((uint32_t) &g_pfnVectors - FLASH_BASE), // start address depends on bootloader size
        MAX_FLASH((min(LL_GetFlashSize(), 512ul) * 1024)) // LL_GetFlashSize returns size in kB. 512 is max for a bank
```
but still, you should be able to get closer to 50%

JAndrassy, 2023-10-24:
The F4 flash is split into sectors. 0 to 3 are 16 kB, 4 is 64 kB and the rest are 128 kB.
ArduinoOTA is set to store the update starting from sector 5. (The first 128 kB sector and half of a 512kB flash).
 
You can change the sector to use with define OTA_STORAGE_STM32_SECTOR in boards.local.txt

Create a file boards.local.txt next to boards.txt file in hardware package. There you can set build.extra_flags for individual boards, like this: 
```
GenF4.menu.pnum.DIYMORE_F407VGT.build..extra_flags=-DOTA_STORAGE_STM32_SECTOR=8 
```

gcharles81, 2023-10-24:
> file in hardware package. There you can set build.extra_flags for individual boards, like this:

Thank you for the answer and the hint but its too complex to do that I do not know the exact locations where the board.txt files are very difficult to find out what exactly I have to do Thanks

JAndrassy, 2023-10-25:
C:\Users\user\AppData\Local\Arduino15\packages\STM32\hardware\stm32\1.9.0

maxsize is the space for the sketch. sector 5 starts at 128 kB  so it is 131072 bytes in sector 0 to 4

gcharles81, 2023-10-26:
Hello I created the board.local.txt file but I still have the same issue that not enough space is available 

[image]

[image]

JAndrassy, 2023-10-27:
did you restart the IDE?

gcharles81, 2023-10-27:
Yes have restarted platformio multiple times ,

JAndrassy, 2023-10-27:
PlatformIO? maybe it has the boards package somewhere else or it doesn't process local.txt files

gcharles81, 2023-10-27:
Thanks for all your help I managed to get it working by adding -DOTA_STORAGE_STM32_SECTOR=9  as a build flag in platformio.ini i attached below for anyone as a reference in future 
[image]

JAndrassy, 2023-10-27:
so yes, Arduino doesn't have a project configuration file. that is better in plaformio

closed 2023-10-27 by JAndrassy