JAndrassy/ArduinoOTA #226: "No monitor available for the port protocol network" on R4-WiFi

flylow, 2023-11-05:
I'm trying a modified version of WiFi101_OTA.ino from the examples directory on my R4 with WiFi (code below). After loading the sketch on my board, waiting a minute for the network port to show in the IDE, and then selecting it, the error  "No monitor available for the port protocol network" is displayed near the IDE's serial monitor window. The sketch runs fine, showing network status in the status monitor, until I switch to the network port and receive the error. 

--- Environment and modified code ----
Windows 10 Pro, and IDE 2.2.1
I have copied platform.local.txt from
  https://github.com/JAndrassy/ArduinoOTA/tree/master/extras/renesas,
  and placed in the IDE's "...hardware\renesas_uno\1.0.5" directory (same
  directory as standard platform.txt).

`#include <WiFiS3.h>
#include <ArduinoOTA.h>

#include "arduino_secrets.h" 

char ssid[] = SECRET_SSID;        // your network SSID (name)
char pass[] = SECRET_PASS;    // your network password 
int status = WL_IDLE_STATUS;     // the WiFi radio's status

void setup() {
  //Initialize serial and wait for port to open:
  Serial.begin(9600);
//  while (!Serial) {
  // wait for serial port to connect. Needed for native USB port only
//}

  // check for the WiFi module:
  if (WiFi.status() == WL_NO_MODULE) {
    Serial.println("Communication with WiFi module failed!");
    // don't continue
    while (true);
  }

  // attempt to connect to WiFi network:
  while (status != WL_CONNECTED) {
    Serial.print("Attempting to connect to WPA SSID: ");
    Serial.println(ssid);
    // Connect to WPA/WPA2 network:
    status = WiFi.begin(ssid, pass);

    // wait 10 second for connection:
    delay(10000);
  }

  // start the WiFi OTA library with internal (flash) based storage
  // The IDE will prompt for this password during upload
  ArduinoOTA.begin(WiFi.localIP(), "Arduino", "password", InternalStorage);

  // you're connected now, so print out the status:
  printWifiStatus();  

}

void loop() {
  // check the network connection once every 5 seconds:
  delay(5000); 

  printWifiStatus(); //sign of life
  // check for WiFi OTA updates
  ArduinoOTA.poll();  
}

void printWifiStatus() {
  // print the SSID of the network you're attached to:
  Serial.print("SSID: ");
  Serial.println(WiFi.SSID());

  // print your WiFi shield's IP address:
  IPAddress ip = WiFi.localIP();
  Serial.print("IP Address: ");
  Serial.println(ip);

  // print the received signal strength:
  long rssi = WiFi.RSSI();
  Serial.print("signal strength (RSSI):");
  Serial.print(rssi);
  Serial.println(" dBm");
}`

JAndrassy, 2023-11-05:
select the com port for Serial Monitor

flylow, 2023-11-05:
While attached via USB, switching from com port to the network port, results in the "no monitor" error. Should I not expect to see the messages as before, just sent over the network connection instead of USB? Or, is the network port only used for the upload?

JAndrassy, 2023-11-06:
Serial Monitor in IDE works only with com port.

you can use my TelnetStream library to OTA monitor with telnet

closed 2023-11-06 by JAndrassy

flylow, 2023-11-06:
Awesome, I'll check it out. Thanks!!