JAndrassy/ArduinoOTA #228: client.get(buff) causes freeMemory to drop significantly, .stop() doesn't release

JEG8, 2023-11-14:
Hi,

I am facing an issue with the <ArduinoHttpClient.h> library (in conjunction with <ArduinoOTA.h> and <WiFiNINA.h>) via your OTASketchDownloadWifi example, where I make a very simple call of client.get(buff), where buff is defined as the path for my server. My freeMemory (between heap and stack) drops by 1.5K when this call happens. I am assuming the memory is being claimed by something in the ArduinoHttpClient library. When I call client.stop(), the memory is not released. This makes it very difficult to use in a system where memory is critical.

Is there any reason why this would happen? Is there a recommended method to prevent this drop or does your library provide a way to fix this?

Thanks for any help in advance.

JAndrassy, 2023-11-14:
is it not some glitch in the code for the free memory calculation?

JEG8, 2023-11-14:
No, it is directly from MemoryFree library. https://github.com/mpflaga/Arduino-MemoryFree/blob/master/MemoryFree.cpp

JAndrassy, 2023-11-14:
I used that too so I know it is OK. Once I had some other and it didn't work good.

Is the buff string terminated by 0?

JEG8, 2023-11-14:
Yes, the buff string is being formulated using the code below. snprintf properly adds the null terminator to the string. I printed out the hex contents and did in fact see that the null terminator.

```
char buff[32];
  snprintf(buff, sizeof(buff), PATH);
```

JAndrassy, 2023-11-14:
The WiFiNINA library allocates 1500 bytes for buffers at once, but only once

JEG8, 2023-11-14:
Great find! Thank you, this explains what I am seeing. Do you know of any way I can save memory with this implementation?

JAndrassy, 2023-11-14:
you can change the WIFI_SOCKET_BUFFER_SIZE at the beginning of WiFiSocketBuffer.cpp

closed 2023-11-14 by JEG8