JAndrassy/ArduinoOTA #230: OTA with Ethernet from a URL

IgnacioSJ, 2023-11-23:
Hi!
I'm trying to do an OTA update from a URL (e.g. an https://raw.githubusercontent.com/... containing a .bin file) with Ethernet connection.
I have seen that this library can do OTA updates with Ethernet but there is no example (or I have not found it) of an update out from a local network. 
Could be possible to do this with ArduinoOTA library?
Thank you!

JAndrassy, 2023-11-23:
https://github.com/JAndrassy/ArduinoOTA/blob/master/examples/Advanced/OTASketchDownload/OTASketchDownload.ino

IgnacioSJ, 2023-11-23:
Hi @JAndrassy!
If I'm using an URL from Internet, what parameters do I have to write on SERVER and SERVER_PORT?

JAndrassy, 2023-11-23:
Ethernet library doesn't support https, but you can try it with SSLClient library.
https port is 443