JAndrassy/ArduinoOTA #234: Support for arduino Opta (STM32H747XI)

JEREDEK, 2023-12-11:
The opta already runs on STM32H747XI, so i thought it would be compatible, but it seems it returns the ` undefined reference to `InternalStorageClass'` error.

Any possibility of adding opta support?

JAndrassy, 2023-12-11:
see the custom storage example and https://github.com/arduino-libraries/ArduinoIoTCloud/blob/master/src/utility/ota/OTA-portenta-h7.cpp

franc3sc0s92, 2024-01-01:
I should also make it work on opta. Could you give me more information?

paulusTecnion, 2024-06-07:
> see the custom storage example and https://github.com/arduino-libraries/ArduinoIoTCloud/blob/master/src/utility/ota/OTA-portenta-h7.cpp

This link is dead. Is there now support for opta or some other example that we can use?

JAndrassy, 2024-06-07:
https://github.com/arduino-libraries/ArduinoIoTCloud/blob/master/src/ota/implementation/OTASTM32H7.cpp

NDBiophotonicsProf, 2024-10-23:
Has anyone gotten this to work? It would be helpful for a project, but I don’t have the expertise to understand the instructions above.  Thanks in advance.