JAndrassy/ArduinoOTA #239: Arduino Mega not showing up in "network ports"

JaspervanKol, 2024-01-11:
A few months back I had burned the bootloader to a few Arduino Mega's and I could upload a sketch using OTA using ethernet. Now I wanted to update the software and I can't see the Arduino's in the "network ports" section. I can see all my ESP's as shown below but not my Arduino's.

[image]

I already tried a new Arduino and burned the bootloader again but also this Arduino is not showing up in the list. Also the example sketch "OTEthernet" does not work, I can see the debug log in the serial that it is connected with an IP address but still not showing up in the "network ports" section.

I am using the following lib's and versions:
Arduino IDE 2.2.1
ArduinoOTA 1.0.12
MyBoards https://github.com/jandrassy/my_boards

Does somebody know how to fix this issue?

JAndrassy, 2024-01-11:
do you have the IDE 2 updates in my_boards?

JaspervanKol, 2024-01-11:
I downloaded my_boards back in late september 2023 and it was working with Arduino IDE 2. To check for sure, I downloaded my_boards again (today 11-01-2024) and burned the bootloader to a new Arduino Mega. But still with the OTA example sketch I don't see it appear in the "network ports"

JaspervanKol, 2024-01-11:
Okay very weird behavior...

I waited for around 5 minutes while doing nothing in the IDE. Then I tried again, and there it was in the "network ports". But after uploading a sketch successfully they were already gone. See picture below of the successful flash but no entries (Arduino Mega) in the "network ports"

[image]

While still connected to the port "192.168.2.184" uploads are possible. But I can't connect to another one (Arduino Mega). After that they didn't show again...

JAndrassy, 2024-01-12:
consider https://github.com/JAndrassy/ArduinoOTA#ota-upload-from-ide-without-network-port

JaspervanKol, 2024-01-13:
I followed the instructions and eddited the "programmers.txt" located in "C:\Users\jaspe\Documents\Arduino\hardware\my_boards\avr" to:

```
arduinoOTA215.name=Arduino OTA (192.168.2.215)
arduinoOTA215.program.tool=arduinoOTA
arduinoOTA215.ip=192.168.2.215

arduinoOTA218.name=Arduino OTA (192.168.2.218)
arduinoOTA218.program.tool=arduinoOTA
arduinoOTA218.ip=192.168.2.218

arduinoOTA237.name=Arduino OTA (192.168.2.237)
arduinoOTA237.program.tool=arduinoOTA
arduinoOTA237.ip=192.168.2.237

arduinoOTA238.name=Arduino OTA (192.168.2.238)
arduinoOTA238.program.tool=arduinoOTA
arduinoOTA238.ip=192.168.2.238

arduinoOTA239.name=Arduino OTA (192.168.2.239)
arduinoOTA239.program.tool=arduinoOTA
arduinoOTA239.ip=192.168.2.239
```

But in de IDE at "programmer" I only see the old entry of the file:

```
Arduino OTA (192.168.1.102)
Arduino OTA (192.168.1.104)
```

When I remove the whole "my_boards" folder, all the entries in "board" are gone but the entries in "programmer" are still there. Is there another file I need to edit or is the IDE not fully updating?

JaspervanKol, 2024-01-13:
I found out that the IDE is caching some files. I deleted the folder "arduino-ide" located in "C:\Users\jaspe\AppData\Roaming". After restarting the IDE all the programmers showed up!

Now I have another problem where it is defaulting to a standard username & password while using "Upload Using Programmer":

```
"C:\Users\jaspe\Documents\Arduino\hardware\my_boards/tools/arduinoOTA.exe" -address 192.168.2.218 -port 65280 -t 30 -username arduino -password password -sketch "C:\Users\jaspe\AppData\Local\Temp\arduino\sketches\864D53D4FCD5ED644FFD1B948E0F06EC/Control_v4.0.ino.bin" -upload /sketch -b
Connecting to board ...  done
Uploading sketch ...  done
Flashing sketch ...  done
Error flashing the sketch:Unauthorized
Failed programming: uploading error: exit status 1
```

Normally it will ask for the password, right?

JAndrassy, 2024-01-13:
sorry, not for programmer. you have to put the password into platform.txt command. maybe it is possible to put it in programmers.txt