JAndrassy/ArduinoOTA #240: R4 Wifi upload fails with "Error flashing the sketch"

MonkaKokosowa, 2024-01-18:
So, after changing library to WiFiS3 in WiFi101_OTA and adding platform file I can't seem to be able to upload to the board via WiFi. It is connected to it without issues and IDE connects to the board (verifying the password) too, but fails the flashing step. 

Only thing visible in the console is: 
```Connecting to board ...  done
Uploading sketch ...  done
Flashing sketch ... Error flashing the sketch
Failed uploading: uploading error: exit status 1
```

Is there any good way to debug this issue?

JAndrassy, 2024-01-18:
do you have verbose mode truned on in IDE Preferences?

could it be timeout of the upload tool? it is only 10 seconds.
newer version of the arduinoOTA tool allows a timeout parameter
https://github.com/arduino/arduinoOTA/releases
replace the 1.3.0 exe and in platform.local.txt add the -t option in seconds

MonkaKokosowa, 2024-01-18:
After setting timeout to 20 it works without issues. I guess my router might be a bit slow. Thanks a lot! <3

closed 2024-01-18 by MonkaKokosowa

JAndrassy, 2024-01-18:
the 10 secs for for AVR Uno for Arduino Yun.
the sketch for Uno R4 is larger and the WiFiS3 library is not the fastes

BergHeisen, 2024-08-11:
> do you have verbose mode enabled in IDE Preferences?
> 
> could it be timeout of the upload tool? it is only 10 seconds. newer version of the arduinoOTA tool allows a timeout parameter https://github.com/arduino/arduinoOTA/releases replace the 1.3.0 exe and in platform.local.txt add the -t option in seconds

Well I did that, and uploading an example sketch like one of the LED Matrix works the first time, but the second time it just goes on indefinetly. I have ArduinoOTA.h and WiFiS3.h included at the top of the file, and upload using the Fake Programmer.

JAndrassy, 2024-08-12:
> I have ArduinoOTA.h and WiFiS3.h included at the top of the file

that is not enough