JAndrassy/ArduinoOTA #247: I need help with Arduino Mega OTA

heinsschonhals, 2024-05-09:
Hello, I am getting the following feedback when I send a sketch via the network port:

"Connecting to board... done
Uploading sketch...done
Flashing sketch... Error flashing the sketch"

I have already changed the timeout but it may not be correct, as this error message comes in approximately 10 seconds, but the following command is given before this message appears:

"C:\Users\heins\AppData\Local\Arduino15\packages\arduino\tools\arduinoOTA\1.3.0/bin/arduinoOTA -address 192.168.3.6 -port 65280 -username arduino -password heins -sketch C:\Users\ heins\AppData\Local\Temp\arduino_build_653848/testeOTA.ino.bin -upload /sketch -b -t 60"

I installed the Megacore bootloader "stk500boot_v2_atmega2560_115200_16000000L.hex", if it's not the correct one, if you could send me the correct bootloader, I've also replaced the arduinoOTA.exe file with version 1.4.1, I think it was in 1.3.0

JAndrassy, 2024-05-09:
you need optiboot botloader. not stl500v2

heinsschonhals, 2024-05-09:
Do you have the file or link, as I've already used some optiboot but they didn't work, until optiboot 8.0 I couldn't find a bootloader for the mega2560

heinsschonhals, 2024-05-10:
> você precisa do botloader optiboot. não stl500v2

ok, I managed to make sending the sketch to the network port work, but I need to restart the Arduino manually for the new sketch to work, something

mcerqueira1509, 2024-05-10:
I got a mega2560 with optiboot (using my_boards),
and the sketch using: ArduinoOTA.begin(Ethernet.localIP(), "arduino", "password", SDStorage);

after I select the remote target, I get:
Sketch uses 28422 bytes (10%) of program storage space. Maximum is 261120 bytes.
Global variables use 1822 bytes (22%) of dynamic memory, leaving 6370 bytes for local variables. Maximum is 8192 bytes.
Connecting to board ...  done
Uploading sketch ...  done
Flashing sketch ...  done
OK
Sketch uploaded successfully

but the board resets and the new sketch is not running.
do you know what is missing?

JAndrassy, 2024-05-10:
and after manual reset the new sketch runs?

heinsschonhals, 2024-05-10:
> e após a reinicialização manual o novo esboço é executado?

yes execute the new sketch

heinsschonhals, 2024-05-10:
> Consegui um mega2560 com optiboot (usando my_boards), e o sketch usando: ArduinoOTA.begin(Ethernet.localIP(), "arduino", "password", SDStorage);
> 
> depois de selecionar o destino remoto, recebo: Sketch usa 28.422 bytes (10%) do espaço de armazenamento do programa. O máximo é 261120 bytes. As variáveis ​​globais usam 1.822 bytes (22%) de memória dinâmica, deixando 6.370 bytes para variáveis ​​locais. O máximo é 8.192 bytes. Conectando à placa... concluído Carregando esboço... concluído Esboço piscando... concluído OK Esboço carregado com sucesso
> 
> mas a placa é reiniciada e o novo esboço não está em execução. você sabe o que está faltando?

I think the optiboot bootloader is missing, which has the copy_flash_pages function

JAndrassy, 2024-05-10:
SDStorage doesn't need copy_flash_pages

heinsschonhals, 2024-05-10:
> SDStorage não precisa de copy_flash_pages.
> 
> muitos relatam esse problema com seu Mega. de alguma forma, ele não reinicia corretamente

I tried to force the use of the watchdog together with the ArduinoOTA.h utility in the function:

"void end() {
 ArduinoOTAClass<NetServer, NetClient>::end();
 mdnsSocket.stop();
 wtd_enable(WDTO_4S);
 }"

still calling the library:

"#include <avr/wdt.h>"

JAndrassy, 2024-05-10:
the ArduinoOTA library already uses wdt to reset the MCU

heinsschonhals, 2024-05-10:
> a biblioteca ArduinoOTA já usa wdt para redefinir o MCU. end() não é usado no processo de atualização. só permite parar a biblioteca.

in which file and which function calls the watchdog, I'm looking for this now

JAndrassy, 2024-05-10:
> in which file and which function calls the watchdog, I'm looking for this now

copy_flash_pages in bootloader

heinsschonhals, 2024-05-10:
> > em qual arquivo e qual função chama o watchdog, estou procurando isso agora
> 
> copy_flash_pages no bootloader

I'm using a set of optiboot 8 cards, maybe that's it, but it shouldn't work when I restart manually, right?

heinsschonhals, 2024-05-10:
I use the following configuration
[image]

JAndrassy, 2024-05-10:
that doesn't have copy_flash_pages

heinsschonhals, 2024-05-10:
> que não tem copy_flash_pages. use a versão MegaCore ou my_boards
> 
> aliás: o "e após a redefinição manual o novo esboço é executado?" a pergunta era para mcerqueira1509

I found it in your repository, later I will test it and give feedback here if it works to help more people

[image]

heinsschonhals, 2024-05-10:
Another question, if you have already tested it, sending the sketch outside the local network, through a public IP, would there be a manual way to add the network port to the IDE?

JAndrassy, 2024-05-10:
yes, read the README

mcerqueira1509, 2024-05-10:
for me, after manual reset, still running the old sketch

JAndrassy, 2024-05-10:
here https://github.com/JAndrassy/ArduinoOTA

peno64, 2024-07-30:
I must be doing something wrong...

I have an arduino mega 2560 R3

I can upload the blink sketch to it and it works with the default bootloader.

I burned the new bootloader from https://github.com/JAndrassy/my_boards/blob/master/avr/bootloaders/optiboot8/optiboot_atmega2560.hex
To be able to do this, I copied this file to C:\Users\peno64\AppData\Local\Arduino15\packages\Optiboot\hardware\avr\0.8.0\bootloaders\optiboot
It says success when I did this.
I appended the content of https://github.com/JAndrassy/my_boards/blob/master/avr/boards.txt to my boards.txt
In the IDE I choose board Optiboot 8.0 - Arduino Mega 2560 (Optiboot)
[image]
Now I upload, still via usb, the blink sketch. Upload hangs some time and then gives an error.

I can burn the default Arduino Mega bootloader again after this and everything is ok again.

What am I doing wrong?

Thank you