JAndrassy/ArduinoOTA #248: Success response but sketch is not replaced using Mega OTA

mcerqueira1509, 2024-05-10:
Hello,
I got a mega2560 with optiboot (using my_boards),
and the sketch using: ArduinoOTA.begin(Ethernet.localIP(), "arduino", "password", SDStorage);

after I select the remote target, I get:
Sketch uses 28422 bytes (10%) of program storage space. Maximum is 261120 bytes.
Global variables use 1822 bytes (22%) of dynamic memory, leaving 6370 bytes for local variables. Maximum is 8192 bytes.
Connecting to board ... done
Uploading sketch ... done
Flashing sketch ... done
OK
Sketch uploaded successfully

but the board resets and the new sketch is not running.
after manual reset board still running the old sketch.

using mega 2560 with ethernet shield (with sd in the ethernet shield)

about the sd bootloader:
before upload the old sketch I used USBasp as programmer 
choose my_boards_avr -> Arduino Mega 2560 (Optiboot)
and burn bootloader

JAndrassy, 2024-05-10:
what is this?
```
ArduinoOTA.begin(Ethernet.localIP(), "arduino", "password", SDStorage);
```

mcerqueira1509, 2024-05-10:
Ok, thank you. So that is what I was not getting right, can you point a reference for a sd bootloader ?

JAndrassy, 2024-05-10:
> with the sd bootloader can I upload sketch using usb cable ?

no. zevero's avr_boot doesn't support USB upload. you can then only put the update on the SD card or OTA upload it

mcerqueira1509, 2024-05-10:
is there a way to work using both?

JAndrassy, 2024-05-10:
no. I attempted to update it, but was not successful. I didn't try really hard because I didn't really need it. I don't have a computer near the project so if OTA fails it is simpler to take the SD card, put it into the computer and copy the update bin.

But I use Atmega1284p which has only half as flash size than Mega. Why you don't use IntenalStorage?

mcerqueira1509, 2024-05-10:
I got it working using internal storage downloading the hex file, just think is easier using the IDE... 
but will consider the trade offs.

JAndrassy, 2024-05-10:
InternalStorage works with upload from IDE too

mcerqueira1509, 2024-05-10:
thanks for the clarifications

closed 2024-05-10 by mcerqueira1509