JAndrassy/ArduinoOTA #250: arduino uno r4 wifi error Serial.println("There is not enough space to store the update. Can't continue with update.");

brsbnkc, 2024-05-24:
I am reaching out regarding an issue I encountered while using the ArduinoOTA library.

I am currently working with an Arduino Uno R4 WiFi board and attempting to implement OTA updates using your library. However, I am encountering an error message stating "There is not enough space to store the update. Can't continue with update." Despite the fact that the downloaded file size (172140 bytes) is smaller than the maximum program storage space on the board (262144 bytes), I am still encountering this issue.

I have thoroughly checked the memory usage and the size of the downloaded file, and everything seems to be in order. I have also reviewed my code for any potential memory-consuming operations but couldn't find any.

Could you please assist me in resolving this issue or provide any insights into what might be causing it? Any help or suggestions would be greatly appreciated.

Thank you very much for your time and assistance.

JAndrassy, 2024-05-24:
the library uses upper half of the flash memory to store the update binary. after the upload the binary is copied to the run location in the lower half of the flash memory

brsbnkc, 2024-05-24:
Could you please provide guidance on how to effectively resolve this issue? Specifically, I would like to know if there are any recommended steps or configurations to ensure sufficient space is available in the upper half of the flash memory during the update process.

JAndrassy, 2024-05-24:
I wonder why is your sketch so large?
If the sketch is bigger than half of the flash, you could use SD card with SDStorage and SDU library.

brsbnkc, 2024-05-24:
My code hosts several libraries and facilitates RESTful API communication. Additionally, I have a device equipped with an HMI (Human-Machine Interface) screen, which contributes to the larger size of the code file. The device comes with an SD card, and I aim to perform updates over the internet. Is it possible to download the file from the internet and update it via the SD card?

Are there any limitations or challenges I should be aware of in this regard? Do you have any suggestions or alternative solutions?
Can you share an example?
Thank you.

JAndrassy, 2024-05-24:
please read the README

brsbnkc, 2024-05-24:
Investigated the file you specified but couldn't find anything helpful. What I want to do is to download a binary file from the internet and copy it to an SD card, and then perform the update process from the file on the SD card. Can I do this? Could you provide an example?

JAndrassy, 2024-05-24:
> In advanced section of examples you can find examples of sketch update over download from a http server. One example shows update over the InternalStorage object of the ArduinoOTA library. The example for update over SD card doesn't use this library at all.

brsbnkc, 2024-05-24:
Then I guess I won't be able to update via OTA using this library.

JAndrassy, 2024-05-24:
this library  was created for upload from IDE. additionally the InternalStorage objects can be useful with other transports if flash is used as storage.
the SD way is supported by the SDU library, which I ported for Uno R4 (and it was a lot of work to make it small)

closed 2024-05-24 by JAndrassy

JAndrassy, 2024-05-24:
SDU library does it on restart. but both sketches must include it.
(and the SD card wiring must match the one required by the SDU library)
https://github.com/arduino/ArduinoCore-renesas/blob/main/libraries/SDU/examples/Usage/Usage.ino