JAndrassy/ArduinoOTA #252: I need help with Arduino Mega OTA upload

yuny0404, 2024-06-21:
First of all, sorry for my bad English.

I am using Arduino Mega 2560, and uploaded optiboot_atmega2560.hex as the bootloader.

I downloaded ArduinoOTA as a zip file and added the library. 

I am trying to upload by selecting Arduino Mega or Mega 2560.

I'm trying to upload a basic Blink example now, but the upload fails.

I wish I could tell you what the reason is

This is an image for reference.

[image]
[image]
[image]

JAndrassy, 2024-06-21:
with Optiboot you have to use my_boards or Mega Core for upload to Mega. The Arduino AVC platform uses upload settings for the standard bootloader

yuny0404, 2024-06-24:
Thanks for your help. Thanks to this, the file upload was successful. 

This time I'm running OTASketchDownload.ino.

When I run it, a 301 error occurs in the serial.

I'll show you the code and serial I use.

Is there anything I need to fix?

The bin file used in the path is here.

Please help me😭😭
https://github.com/dalgona-edu/OTA_test/blob/main/OTA_update_check.ino.bin

### OTASketchDownload.ino
```
/*

 This example downloads sketch update over network.
 It doesn't start the OTA upload sever of the ArduinoOTA library,
 it only uses the InternalStorage object of the library
 to store and apply the downloaded binary file.

 To create the bin file for update of a SAMD board (except of M0),
 use in Arduino IDE command "Export compiled binary".
 To create a bin file for AVR boards see the instructions in README.MD.
 To try this example, you should have a web server where you put
 the binary update.
 Modify the constants below to match your configuration.

 Created for ArduinoOTA library in February 2020
 by Juraj Andrassy
 */

#include <Ethernet.h>
#include <ArduinoHttpClient.h>

#define NO_OTA_NETWORK
#include <ArduinoOTA.h> // only for InternalStorage

const short VERSION = 1;

byte mac[] = { 0x54, 0x10, 0xEC, 0xA0, 0xEC, 0x43 };

#ifdef ARDUINO_SAM_ZERO // M0
#define Serial SerialUSB
#endif

void handleSketchDownload() {

  const char* SERVER = "github.com"; // must be string for HttpClient
  const unsigned short SERVER_PORT = 80;
  const char* PATH = "/dalgona-edu/OTA_test/blob/main/OTA_update_check.ino.bin";
  const unsigned long CHECK_INTERVAL = 5000;

  static unsigned long previousMillis;

  unsigned long currentMillis = millis();

  if (currentMillis - previousMillis < CHECK_INTERVAL)
    return;
  previousMillis = currentMillis;

  EthernetClient transport;
  HttpClient client(transport, SERVER, SERVER_PORT);

  char buff[32];
  snprintf(buff, sizeof(buff), PATH, VERSION + 1);

  Serial.print("Check for update file ");
  Serial.println(buff);

  client.get(buff);

  int statusCode = client.responseStatusCode();
  Serial.print("Update status code: ");
  Serial.println(statusCode);
  if (statusCode != 200) {
    client.stop();
    return;
  }

  long length = client.contentLength();
  if (length == HttpClient::kNoContentLengthHeader) {
    client.stop();
    Serial.println("Server didn't provide Content-length header. Can't continue with update.");
    return;
  }
  Serial.print("Server returned update file of size ");
  Serial.print(length);
  Serial.println(" bytes");

  if (!InternalStorage.open(length)) {
    client.stop();
    Serial.println("There is not enough space to store the update. Can't continue with update.");
    return;
  }
  byte b;
  while (length > 0) {
    if (!client.readBytes(&b, 1)) // reading a byte with timeout
      break;
    InternalStorage.write(b);
    length--;
  }
  InternalStorage.close();
  client.stop();
  if (length > 0) {
    Serial.print("Timeout downloading update file at ");
    Serial.print(length);
    Serial.println(" bytes. Can't continue with update.");
    return;
  }

  Serial.println("Sketch update apply and reset.");
  Serial.flush();
  InternalStorage.apply(); // this doesn't return
}

void setup() {

  Serial.begin(115200);
  while (!Serial);

  Serial.print("Sketch version ");
  Serial.println(VERSION);

  Serial.println("Initialize Ethernet with DHCP:");
  if (Ethernet.begin(mac) == 0) {
    Serial.println("Failed to configure Ethernet using DHCP");
    while (true);
  }
  Serial.println("Ethernet connected");
}

void loop() {
  // check for updates
  handleSketchDownload();

  // add your normal loop code below ...
}
```

### Serial
```
Sketch version 1
Initialize Ethernet with DHCP:
Ethernet connected
Check for update file /dalgona-edu/OTA_test/blob/main
Update status code: 301
Check for update file /dalgona-edu/OTA_test/blob/main
Update status code: 301
Check for update file /dalgona-edu/OTA_test/blob/main
Update status code: 301
Check for update file /dalgona-edu/OTA_test/blob/main
Update status code: 301
Check for update file /dalgona-edu/OTA_test/blob/main
Update status code: 301
```

JAndrassy, 2024-06-24:
the Ethernet library doesn't support secure connection (https)

JAndrassy, 2024-06-24:
WiFiNINA library is not supported on Arduino Mega

yuny0404, 2024-06-24:
If so, you should use the http download link in the OTASketchDownload.ino example.

Thanks for your help😊😊

I'll test it and come back.

yuny0404, 2024-06-24:
hmm..

I tried several times to insert a link in http format.

However, when I created the link, only the https format link was created.

How can I create a link in http format?

I hope it isn't too difficult...

JAndrassy, 2024-06-24:
most public sites don't allow insecure http

yuny0404, 2024-06-24:
It would be great if https format was supported. 

So would it be better to use a local server?

If you know of any other method than this, please share😊😊

JAndrassy, 2024-06-24:
other way in which way? I don't know what arduino libraries are still able to connect to public sites. many fail on newer versions of the secure protocol. my WiFiEspAT library with Jiri Bilek's firmware in esp8266 may be able to connect to public sites. Or with an ESP32 with AT firmware.

yuny0404, 2024-06-25:
Oh... I can't change from mega2560 to another board.

And you can't use SD cards either.

So how can I use the OTA function?

JAndrassy, 2024-06-25:
WiFiEspAT is for esp8266 or esp32 as WiFi adapter.

yuny0404, 2024-06-25:
Thank you so much for letting me know, but I definitely have to complete the OTA function on the mega2560.

It's okay if you don't use WiFiEspAT.

Feel free to use other examples.

Is there any way to do OTA on mega2560?

I really want this to succeed😭😭

JAndrassy, 2024-06-25:
so upload from IDE works. you have to code other transports in your sketch. you can do a download, but for https sites you have to use a network adapter which handles the encryption, because Mega doesn't have the computing power to do the encryption.
I don't understand why you repeat the you have to do it on Mega. I never suggested that you should not.

yuny0404, 2024-06-25:
The reason I said that I had to use mega was because I couldn't use ESP32.

This is how I understood it. 
Since WiFiEspAT can only be used in ESP, I understood that the WiFiEspAT example cannot be used in MEGA.

Maybe I'm misunderstanding something?

Anyway, I have no choice but to use mega, and I have to use OTA on mega.

Whatever method you use to do OTA on mega is fine. 

I would be grateful if you could tell me how to do it.

JAndrassy, 2024-06-25:
WiFiEspAT is used in anything else than esp to communicate wit the AT firmware in esp

yuny0404, 2024-06-25:
I read Read.me.

Let me explain my situation now.

1. I can only use mega2560 now. ESP cannot be used

2. I uploaded the optiboot bootloader to mega.

3. I set the IDE to use Arduino Mega or Mega 2560 (Optiboot).

4. I tried testing OTA with the OTASketchDownload.ino example.

5. There was a problem that links in https format could not be used.

What I'm curious about here is this

Q1. Can I create a link in http format? If you can make it, how can you make it?

Q2. Is there a way to test OTA on mega with an example other than the OTASketchDownload.ino example?

Sorry for the many questions.
But I really want to succeed at this.

JAndrassy, 2024-06-25:
why did you start with an advanced example? 
with Ethernet library you should use the OTEthernet example and upload from the IDE

yuny0404, 2024-06-25:
Oh, I'm sorry. I have already tested the OTEthernet example. 
It worked well without any problems.

JAndrassy, 2024-06-25:
ok. then my job is done. the ArduinoOTA library works