JAndrassy/ArduinoOTA #261: Firmware updating doesn't works with huge partition table

Vano3d, 2024-10-31:
Hello,
I'm using huge partition csv:
# Name, Type, SubType, Offset, Size, Flags
nvs, data, nvs, 0x9000, 0x5000,
otadata, data, ota, 0xe000, 0x2000,
app0, app, ota_0, 0x10000, 0x300000,
spiffs, data, spiffs, 0x310000,0xF0000,

ElegantOTA stops updating on 14%, then ESP32 rebooting ((
With standard partition or without updating working perfectly, but I have a big project.

Thanks for assistance!

JAndrassy, 2024-11-01:
I guess you use the esp32 core's ArduinoOTA library and not this one

Vano3d, 2024-11-01:
I use much more libs

gyverlibs/GyverOLED@^1.6.1
	gyverlibs/EncButton@^3.6.0
	gyverlibs/GyverTimer@^3.2
	gyverlibs/GyverBlinker@^1.1.0
	bblanchon/ArduinoJson@^7.0.0
	mathieucarbou/AsyncTCP @ 3.2.10
	https://github.com/ayushsharma82/ElegantOTA#3.1.6
	adafruit/Adafruit ADS1X15@^2.5.0
	bodmer/TFT_eSPI@^2.5.43
	gyverlibs/GyverSegment@^1.4.5
build_flags = -DELEGANTOTA_USE_ASYNC_WEBSERVER=1

JAndrassy, 2024-11-01:
so you use ElegantOTA.  you report the issue in wrong repository

closed 2024-11-01 by JAndrassy

Vano3d, 2024-11-01:
I used ElegantOTA normally until changed partition.csv
I must look for another cause?

JAndrassy, 2024-11-01:
why do you ask here? ask there https://github.com/ayushsharma82/ElegantOTA

Vano3d, 2024-11-01:
OMG, I confused )