JAndrassy/ArduinoOTA #270: segmentation fault when attempting to upload example to Uno R4 WiFi

alan16R, 2025-08-29:
Hi

I'm trying out Arduino OTA on a Uno R4 WiFi. I'm using the example `examples/WiFi101_OTA/WiFi101_OTA.ino` but with the WiFiNINA line replaced with WiFiS3 as below:-

```
 
#include <SPI.h>
#include <WiFiS3.h>
#include <ArduinoOTA.h>

#include "arduino_secrets.h" 
///////please enter your sensitive data in the Secret tab/arduino_secrets.h
/////// Wifi Settings ///////
char ssid[] = SECRET_SSID;      // your network SSID (name)
char pass[] = SECRET_PASS;   // your network password

int status = WL_IDLE_STATUS;
```

I've copied the platform.local.txt into the same folder as platform.txt and deleted the library OTAUpdate from the libraries folder. When I try and upload the sketch using the USB cable, a dialog pops up asking for a password and I give my WiFI password, then I get:-

```
Sketch uses 75516 bytes (28%) of program storage space. Maximum is 262144 bytes.
Global variables use 10464 bytes (31%) of dynamic memory, leaving 22304 bytes for local variables. Maximum is 32768 bytes.
Failed uploading: uploading error: signal: segmentation fault

```

What am I doing wrong?

Thanks in advance!

alan16R, 2025-08-29:
Sorry, should have added:-

1. Using Arduino R4 1.5.1
2. IDE 2.3.5
3. Mac M2

JAndrassy, 2025-08-29:
I don't use the new IDE. Does it ask for the password for the arduino_secrets.h file automatically? Is the arduino_secrets.h opened as tab in IDE?